miguelgrinberg/microdot #175: microdot hangs when aborting static file serve

Tobinator32, 2023-10-24:
Im currently developing a small web app and encountering some issues. im running latest micropython build  with microdot as a precompiled module (not frozen but tried that no difference). There is one websocket and a static file serve. Everything works fine but if i leave the currently loading page in my browser by either pressing that back button or closing the window the webserver freezes. This is only if somehow aborting the currently sending file or pending files that are requested and the connection are still in the waiting queue.
After a short deepdive microdot_async waits for wait_io_event residing inside the asyncio lib. Also the websocket stops initiating.
My question is how can it be that esp32 becomes numb in this scenarios and how to solve the issue. 
Kind Regards Tobi

Tobinator32, 2023-10-24:
Hello,
Thanks for the fast response. Let me explain how it is structured. At first the html is served. This requires some js and css along with some small pictures and a websocket finally. If im aborting the site loading, as the back button appears faster than the whole js and css loading takes a new request for all files will be sent but out of an other html file while the previous ones arent served and not needed anymore. If i make this 3 time or so in a row then the server hangs and doesnt accept new connections.
I could gather up an example this evening/tomorrow throughout the day.
Kind Regards Tobi

miguelgrinberg, 2023-10-24:
If you could provide an example that I can use to reproduce the problem that would help, because you are probably omitting some details. Like for example, when you press the back button what page is it being requested? You must have had some other page in the history.

Tobinator32, 2023-10-25:
Im currently making a little example. Its not demanding enough yet to reproduce the issue though. 
Regarding your suggestion. The backbutton in the browser and the backbutton embedded into the html dont work both. I can jump around the page for quite a bit and then suddenly it stops responding. I can even try to connect with some other client and there no requests are being served as well. Sometimes the whole wifi comes down and a hard reset is required.
As said im on it to provide a good example and im looking forward on finishing it tomorrow through out the day.
Kind Regards and Thanks for the patience

Tobinator32, 2023-10-25:
There it is https://github.com/Tobinator32/microdot_crash_demo. just upload everything onto the esp32( ive precompiled all microdot files in my setup). Now most pages try to load 4 pictures which simulates some traffic doing file requests. Rather quickly it suddenly stops serving any requests. 
If youre encountering any problems with the demo just contact me. 
Kind regards Tobi

Tobinator32, 2023-10-27:
im already a bit further down the road: see here: https://github.com/micropython/micropython/issues/12819

miguelgrinberg, 2023-11-02:
Sounds like this is a bug in MicroPython, so I'm closing for now. Feel free to reopen if there are new developments.

closed 2023-11-02 by miguelgrinberg