miguelgrinberg/microdot #179: Websocket example doesn't work on CPython / Linux: invalid start byte

wohltat, 2023-11-08:
When trying to run the 'echo.py' websocket example i get the folloing error when accessing the server via a browser:
```
> python echo.py 
Traceback (most recent call last):
  File "/home/user/Documents/projects/rfid/websockets_python/microdot/examples/websocket/temp/microdot.py", line 1166, in handle_request
    req = Request.create(self, stream, addr, sock)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Documents/projects/rfid/websockets_python/microdot/examples/websocket/temp/microdot.py", line 384, in create
    line = Request._safe_readline(client_stream).strip().decode()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xfc in position 8: invalid start byte
Traceback (most recent call last):
  File "/home/user/Documents/projects/rfid/websockets_python/microdot/examples/websocket/temp/microdot.py", line 1166, in handle_request
    req = Request.create(self, stream, addr, sock)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Documents/projects/rfid/websockets_python/microdot/examples/websocket/temp/microdot.py", line 384, in create
    line = Request._safe_readline(client_stream).strip().decode()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xfc in position 8: invalid start byte
```

No Webpage is displayed.

miguelgrinberg, 2023-11-08:
Try `http`, not `https`.

wohltat, 2023-11-08:
Ok, Thanks. I missed that one. 
I was about to write that it works on another machine.

The error code didn't really give a hint on the URL typo.

closed 2023-11-08 by wohltat