miguelgrinberg/microdot #195: Hang on python 3.10.4

sophiedegran, 2024-01-07:
Migrate from microdot 1.33 to 2.02
Use Microdot and Websocket package.
It seem to be an asyncio problem..
microdot.py line 25 in invoke handler
TypeError: BaseEventLoop.run_in_executor () got an unexpected keyword argument 'path'

The same code (with microdot 2.02) work perfectly well under micropython 1.21

Perharps the 2.x version is no more compatible directly with python?
I will try with python last release (3.12.x)

miguelgrinberg, 2024-01-07:
This is actually a bug, I'll take care of it shortly.

closed 2024-01-07 by miguelgrinberg

miguelgrinberg, 2024-01-07:
@sophiedegran Please upgrade to 2.0.3 for the fix.

sophiedegran, 2024-01-07:
Thanks
I will continue testing under micropython
It seem more robust and reactiv

Take your time to appreciate 2024!
What you've done is amazing ☺

commit 6712c474 (2024-01-07): Pass keyword arguments to thread executor in the correct way (Fixes #195)