miguelgrinberg/microdot #201: Client ECONNRESET errors when POSTing to micropython microdot from micropython aiohttp

ned-pcs, 2024-01-18:
**Describe the bug**
When using the Unix Micropython to run a server written with version 2.0.3 of microdot, micropython clients (either Unix or ESP32) using `aiohttp` get `ECONNRESET` errors after a POST call. This does not happen when either end is running desktop Python3.

**To Reproduce**
1. get a current build of the Unix Micropython.
2. install version 2.0.3 of microdot in its library
3. run the attached `test_aiohttp_server.py` program with that micropython
4. run the attached `test_aiohttp.py` client program with either that micropython or on an ESP32 board
5. observe the output of both programs

**Expected behavior**
I expect to see no errors.

**Observed behavior**
On the client side, I get this:
```
% tools/micropython test_aiohttp.py
Traceback (most recent call last):
  File "test_aiohttp.py", line 16, in <module>
  File "asyncio/core.py", line 1, in run
  File "asyncio/core.py", line 1, in run_until_complete
  File "asyncio/core.py", line 1, in run_until_complete
  File "test_aiohttp.py", line 11, in main
  File "aiohttp/__init__.py", line 41, in text
  File "aiohttp/__init__.py", line 38, in read
  File "asyncio/stream.py", line 1, in read
OSError: [Errno 104] ECONNRESET
```

**Additional context**
```
% tools/micropython -v       
MicroPython v1.23.0-preview.48.g1da65eed3 on 2024-01-18; linux [GCC 11.4.0] version
```
[aiohttp](https://github.com/micropython/micropython-lib/tree/7cdf70881519c73667efbc4a61a04d9c1a49babb/python-ecosys/aiohttp)

A wireshark capture shows the server side sending a TCP packet with RST/ACK flags set immediately after sending the response. This does not happen when either end is running desktop Python3.  This may indicate that the client side has closed its packet too soon.

Here are the test programs: [test_aiohttp.zip](https://github.com/miguelgrinberg/microdot/files/13982396/test_aiohttp.zip)

miguelgrinberg, 2024-01-18:
Looks like a bug in MicroPython's asyncio library to me. See https://github.com/micropython/micropython-lib/issues/788 for my analysis.

ned-pcs, 2024-02-12:
Closing because it seems to have been fixed.

closed 2024-02-12 by ned-pcs