miguelgrinberg/microdot #209: ".." in URL path resolves correctly, "//" does not

sblive, 2024-02-19:
rules like
`@app.route("/api/v1/system/status", methods=["GET"])`
get correctly matched by e.g.
/api/v1/system/status/../status
/api/v1/system/blahblahdoesnotexist/../status
but not by
/api/v1/system/status//status

miguelgrinberg, 2024-02-20:
I'm confused. None of the examples that you have showed above match the example route that you are using, and that is actually correct. Your route doesn't have any dynamic segments, so only the same exact URL will match. This is true before and after the change I've made yesterday.

```python
>>> p = URLPattern('/api/v1/system/status')
>>> p.match('/api/v1/system/status/../status') is not None
False
>>> p.match('/api/v1/system/blahblahdoesnotexist/../status') is not None
False
>>> p.match('/api/v1/system//status') is not None
False
>>> p.match('/api/v1/system///status')  is not None
False
>>> p.match('/api/v1/system/status') is not None
True
```

sblive, 2024-02-20:
that's weird, I just retested and still this works:
/api/v1/system/asdf/../status
which I believe to be correct and all (?) webservers support.
for multi slashes (which most webservers support) the easy fix would be to replace all multiple slashes in path with single slash.
maybe you don't think it's a bug, I guess it's an interpretation thing and not that important ;)

miguelgrinberg, 2024-02-20:
@sblive How are you testing this exactly?

sblive, 2024-02-20:
I'm sorry, you are right. it seems wget (which I used) and browsers already translate this, so I realized that microdot will never see those URLs. sorry for the confusion!

closed 2024-02-20 by sblive