miguelgrinberg/microdot #212: Integer-type dynamic path component does not match 0

kevin-pololu, 2024-03-06:
**Describe the bug**
An integer-type dynamic path component does not match the value 0.

**To Reproduce**
Steps to reproduce the behavior:
1. Define a route for a URL like `@app.route('/foo/<int:bar>')`
2. Request a URL like /foo/0
3. The response will be a 404

**Expected behavior**
The `<int:bar>` dynamic component should match the value 0.

**Additional context**
This appears similar to #137 but probably has a different cause. I think the problem is in [this check](https://github.com/miguelgrinberg/microdot/blob/901f4e55b83cba4eb297e070ef159ff19d456059/src/microdot/microdot.py#L865); I kind of suspect it isn't even necessary because you've already checked `if arg is None:` a few lines up, but I'm not sure.

Also, similar to what #137 pointed out, the unit tests don't seem to test a value of 0.

miguelgrinberg, 2024-03-06:
Thanks, I'll take care of it.

miguelgrinberg, 2024-03-06:
@kevin-pololu Can I ask you to retest using the main branch? Let me know if there are any remaining issues.

closed 2024-03-06 by miguelgrinberg

kevin-pololu, 2024-03-07:
Seems good now. Thanks!

commit d0a4cf8f (2024-03-06): Handle 0 as an integer argument (Fixes #212)