miguelgrinberg/microdot #214: CircuitPython support

theboyknowsclass, 2024-03-08:
Any chance this will work in CircuitPython?  

There are some Raspberry Pi Pico functions still lacking from micropython :(

miguelgrinberg, 2024-03-08:
It works on CircuitPython, but only on their 9.x pre-releases. They have some bugs that prevent Microdot from working on the 8.x and older versions that I reported to them, and my understanding is that these are only fixed on the main branch, towards the 9.x release.

JimmyDale, 2024-04-03:
I am still having a problem using under circuitpython:

code.py output:
Starting async server on 0.0.0.0:5000...
Traceback (most recent call last):
  File "code.py", line 33, in <module>
  File "/lib/microdot/microdot.py", line 1282, in run
  File "/lib/asyncio/core.py", line 317, in run
  File "/lib/asyncio/core.py", line 276, in run_until_complete
  File "/lib/asyncio/core.py", line 261, in run_until_complete
  File "/lib/microdot/microdot.py", line 1235, in start_server
  File "/lib/asyncio/stream.py", line 231, in start_server
ImportError: no module named 'usocket'

Code done running.

Adafruit CircuitPython 9.1.0-beta.0 on 2024-03-28; Arduino Nano RP2040 Connect with rp2040

miguelgrinberg, 2024-04-03:
I'm sorry, you are correct. I have missed that they do not have a `usocket` class. Unfortunately there isn't much I can do, they need to fix their asyncio module to support creating a server. See https://github.com/adafruit/Adafruit_CircuitPython_asyncio/issues/35.

amadensor, 2024-05-30:
The reason I was considering CircuitPython is that the Pico support for SD cards is gone from Micropython.  It's just SPI.   Might be easier to get that working than the other way around.

mattytrentini, 2024-05-31:
> The reason I was considering CircuitPython is that the Pico support for SD cards is gone from Micropython. It's just SPI. Might be easier to get that working than the other way around.

I haven't used it for a while but I believe SDCard support is available for all ports? Just needs to be mip-installed (or copied) to your device:

https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/storage/sdcard/sdcard.py

Let me know if that's not working for you...

amadensor, 2024-05-31:
You are right, it got moved from core to the separate -lib repo, but only for some ports.   It is still in core on ESP-32.   Thanks for the heads up.  I will be building a web configured combat robot timer that the 4 boxes talk to each other via web services API.   At least that's the plan, because UART required too many dedicated ports.  I will put this little thing through its paces, and find ways to make it better.   It will run on ESP32-WROOM or Pi Pico.   Not 100% sure yet.   The SD card is just a way to get network setup in there without needing to have USB connections and uncommon PC software.

closed 2026-08-29 by miguelgrinberg