miguelgrinberg/microdot #280: SSE not streaming properly in Safari

maxi07, 2025-01-20:
I am testing the functionality of SSE and copied your SSE example 1:1. I can get in running properly with Google Chrome, see here:

https://github.com/user-attachments/assets/524a77d2-d6de-4bb4-a2ff-75ebbc23def0

But when calling the same url with Safari (Version 18.2 (20620.1.16.11.8)), Safari would only render the final result after the loop had finished:

https://github.com/user-attachments/assets/ab5d8b9d-d794-400e-b1b2-75b9ec87681e

maxi07, 2025-01-20:
Something to add: If being handled by some js, it works fine. But it's not directly rendered like in Chrome. Maybe it's helpful if the documentation would include a short example of how this can be handled by js.

miguelgrinberg, 2025-01-20:
SSEs are always supposed to be handled by JS, opening an SSE endpoint as a page is not an intended use, so I wouldn't call this a Safari bug.

Handling SSE from the client side is out of scope for Microdot, which only concerns itself with the server, so I'm not going to include client-side documentation in this project. Users interested in implementing SSE in the front end can consult the appropriate documentation, such as https://developer.mozilla.org/en-US/docs/Web/API/Server-sent_events/Using_server-sent_events.

maxi07, 2025-01-20:
I See - I thought this is worth noting in the docs. If someone would purely copy and paste your example, it would not "work" properly. All other examples provided by your documentation can be pasted and it just works.
But thanks for the clarification, can be closed.

PR #281 (2025-01-22): add js to sse example
This adds some JavaScript examples to the SSE documentation, so the log output is being streamed correctly in Safari. Before it would just show the finished result on Safari, now the progress can be seen in the console.
closes #280
closed 2025-01-22 by miguelgrinberg