miguelgrinberg/microdot #295: Cannot access the web server

pem120, 2025-04-30:
**Describe the bug**
Ther server cannot be accessed via the web browser

**To Reproduce**
Steps to reproduce the behavior:
1. use code:
```import dht
from microdot import Microdot, redirect, send_file

import machine

app = Microdot()
dht = dht.DHT11(machine.Pin(23))

@app.route("/",methods=['GET'])
async def get_data(request):
    if request.method == "GET":
        dht.measure()
        #return f"temp:{dht.temperature()},humi:{dht.humidity()}"
        return "test"

app.run(host="0.0.0.0", port=80, debug=True)
````
**Expected behavior**
The web page should show test

**Additional context**
Using micropython 1.24.1 on a ESP32-wroom-32

pem120, 2025-05-01:
Here is the boot file
```
# This file is executed on every boot (including wake-boot from deepsleep)
import esp

esp.osdebug(None)

import network
import time
wlan = network.WLAN(network.STA_IF)
wlan.active(True)

    
if not wlan.isconnected():
    print('connecting to network...')
    
    time.sleep(1)
    while not wlan.isconnected():
        try:
            wlan.connect("D*****","B*****")
        except:
            pass
        print(wlan.scan())
        time.sleep(5)
print('network config:', wlan.ifconfig())

#import webrepl
#webrepl.start()
import uftpd

pem120, 2025-05-01:
I was testing today when I got this error
```
Traceback (most recent call last):
  File "<stdin>", line 18, in <module>
  File "/lib/microdot/microdot.py", line 1280, in run
  File "asyncio/core.py", line 1, in run
  File "asyncio/core.py", line 1, in run_until_complete
  File "asyncio/core.py", line 1, in wait_io_event
OSError: [Errno 5] EIO
>>> 
```

miguelgrinberg, 2025-05-01:
Errno 5 is a fairly generic error code that can happen for many different reasons, but this indicates that something in your device is not working well.

If you are 100% sure that the networking initialization that you have in boot.py is working and giving you a connection to your router, then my next suggestion I think it would be to flash your device and set it up from scratch.

pem120, 2025-05-01:
Already reflashed the ESP32 but I was able to get curl to connect to it once but after that nothing

miguelgrinberg, 2025-05-01:
I'm not sure how can I help debug this. Maybe try another device? If you have one device that works and one that doesn't then maybe you have a bad device.

pem120, 2025-05-01:
But I used an example from the internet, and it works perfectly fine. Also uftpd also works perfectly and i used it to back up the files before flashing

pem120, 2025-05-01:
I have tried it on an another esp32 and it works perfectly

pem120, 2025-05-01:
I got to work by erasing the flash and then flashing micropython

closed 2025-05-01 by pem120